EverythingSmartHome/everything-presence-one #250: co2 sensor

hugo1906, 2025-03-21:
I can't get values ​​from the co2 sensor even after updating and calibrating

EverythingSmartHome, 2025-03-21:
Most likely reason is you need to change the hardware revision you selected during the flash. Try choosing the opposite

EverythingSmartHome, 2025-03-21:
Yes, did you try selecting rev 1.5?

EverythingSmartHome, 2025-04-03:
Can you please upload startup logs?

hugo1906, 2025-04-10:
where do I find this

daedalist, 2025-06-25:
Can I jump on this? I have the same symptoms:

- EPOne board rev 1.3
- SEN0395
- CO2 module
- BT Proxy enabled
- Stable Firmware

CO2 entity in HA is "Unknown" and doesn't change after pressing "Calibrate". Attaching log file, but the relevant entries seem to be

```
[14:40:34][E][scd4x:035]: Failed to stop measurements
[14:40:34][E][component:138]: Component scd4x.sensor was marked as failed.
[14:40:34][E][component:195]: Component scd4x.sensor set Error flag: unspecified
...
[14:40:42][E][component:088]:   Component scd4x.sensor is marked FAILED: 
```

[logs.txt](https://github.com/user-attachments/files/20896613/logs.txt)

daedalist, 2025-06-25:
I also have a EPOne and CO2 unit that I haven't connected or configured yet, if that helps

EverythingSmartHome, 2025-06-25:
> I also have a EPOne and CO2 unit that I haven't connected or configured yet, if that helps

Can you flash from the user guide here: https://docs.everythingsmart.io/s/products/doc/restore-or-update-your-epl-to-factory-settings-using-a-computer-orj7PBBOJZ

Just updated to show much more logging, the lines you have at the moment aren't really relevant (that line about it being failed can just be a red herring)

daedalist, 2025-06-28:
Actually, it turns out that I misunderstood the alignment of the GPIO pins on the rev 1.3 board -- the labels are on the back of the board and too small for my ageing eyes. Rotated the sensor 90° clockwise, and it all works fine now.

Thanks for your help!