EverythingSmartHome/everything-presence-one #272: mmWave endlessly triggers

VACIndustries, 2025-06-30:
This was not an issue originally when I plugged this in and had it all setup but recently the sensor seems to endlessly trigger on and off.  Even with the sensitives turned down to zero it still occurs.  Wondering if my sensor is defective/died?  I don't think I've even had the device a year at this point.

It's so bad (triggering on and off) that the UI/system hangs/freezes when I click the entity to review the history of the mmWave.

Nothing has changed in the environment either; it is drilled into the wall and hasn't moved and no new major items (moving or not) have been introduced into the surrounding environment.

EverythingSmartHome, 2025-06-30:
Can you post your full settings? I can make suggestions that way

VACIndustries, 2025-06-30:
EP1 mmWave Max Distance - 6.0m
 EP1 mmWave Minimum Distance - 0.6m 
 EP1 mmWave Mode - Presence Detection
 EP1 mmWave off latency - 1s
 EP1 mmWave on latency - 2.0s
 EP1 mmWave Sustain Sensitivity - 0
 EP1 mmWave Threshold Factor - 20
 EP1 mmWave Trigger Distance 6.0m
 EP1 mmWave Trigger Sensitivity - 0

closed 2026-05-11 by EverythingSmartHome