EverythingSmartHome/everything-presence-one #278: Feature request - EPS Reboot

ef1970, 2025-08-05:
EPL has a EPS Reboot option. Any chance that could be added to EP1 as well?

ef1970, 2025-09-05:
Any thoughts?

EverythingSmartHome, 2025-11-07:
Yes will add

> Hi, I would like to have this feature as well.
> 
> Both the EP1 and EPL for me have this issue where it gets stuck on a occupancy status until I reboot the device. For the EPL I can do it easily with the Reboot button, but for the EP1 this is non-existent. Help?

It's very unlikely (though not impossible) they are getting stuck, and far more likely they are detecting something you are not aware is moving that is being accurately detected. Rebooting seems like it fixes the issue as there is different detection sensitivities for initial trigger compared to sustaining a detection but it's not actually fixing it if that makes sense. Jump on Discord or drop us a message and happy to help you tune

ef1970, 2025-11-07:
> It's very unlikely (though not impossible) they are getting stuck, and far more likely they are detecting something you are not aware is moving that is being accurately detected. Rebooting seems [...]

Same thing happens here, sometimes they simply drop off the map, other times they become "stuck", but seemingly working as far as being visible in HA goes.

I have an automation running that reboots them at intervals, and that has "fixed" the problem, although I have not been able to see a pattern what causes some of them to do this.