EverythingSmartHome/everything-presence-one #279: Changing WiFi network without connecting cable

nh2, 2025-08-10:
Hi, I'd be great to be able to change the WiFi network to another one without having to physically connect the sensor via USB.

Most WiFi access points allow changing the network settings remotely, so it's a bit annoying to have devices that require physical rewiring to adjust to it.

I image this to be especially important for rollouts to a large amount of devices, e.g. in an office with 100s of sensors. (In my case it's because I wanted to make my own WiFi name clearer, with only 1 EP1 so far.)

Tangentially related issue for the Everything Presence Lite:

* https://github.com/EverythingSmartHome/everything-presence-lite/issues/298

EverythingSmartHome, 2025-08-10:
Hello! 

You actually don't need to plug it in to USB, you can press and hold the boot button on the board (remove front cover) for 5s till it flashes repeatedly, release and the WiFi settings will be reset. Then if your using Bluetooth provisioning it should just pop up automatically inside of Home Assistant.

nh2, 2025-08-13:
I made a mistake and held the RESET button pressed instead of BOOT.

I now get that the `RUN` LED is flashing, and `dmesg` shows that the device is periodically connecting and disconnecting via USB:

```
[30531.678583] usb 1-4: new full-speed USB device number 112 using xhci_hcd
[30531.802801] usb 1-4: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.33
[30531.802819] usb 1-4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[30531.802828] usb 1-4: Product: USB Serial
[30531.808665] ch341 1-4:1.0: ch341-uart converter detected
[30531.809278] usb 1-4: ch341-uart converter now attached to ttyUSB0

[30535.718406] usb 1-4: USB disconnect, device number 112
[30535.719174] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[30535.719209] ch341 1-4:1.0: device disconnected

[30537.923496] usb 1-4: new full-speed USB device number 113 using xhci_hcd
[30538.048528] usb 1-4: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.33
[30538.048540] usb 1-4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[30538.048546] usb 1-4: Product: USB Serial
[30538.051498] ch341 1-4:1.0: ch341-uart converter detected
[30538.052102] usb 1-4: ch341-uart converter now attached to ttyUSB0

[30541.868970] usb 1-4: USB disconnect, device number 113
[30541.869513] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[30541.869585] ch341 1-4:1.0: device disconnected

[30544.077516] usb 1-4: new full-speed USB device number 114 using xhci_hcd
[30544.201660] usb 1-4: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.33
[30544.201678] usb 1-4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[30544.201686] usb 1-4: Product: USB Serial
[30544.205829] ch341 1-4:1.0: ch341-uart converter detected
[30544.207499] usb 1-4: ch341-uart converter now attached to ttyUSB0

[30548.087304] usb 1-4: USB disconnect, device number 114
[30548.088617] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[30548.088690] ch341 1-4:1.0: device disconnected
```

I'm not sure if this was actually caused by me pressing that button, or happened already before.

Because today I got the problem of 

* https://github.com/EverythingSmartHome/everything-presence-one/issues/64
* https://github.com/EverythingSmartHome/everything-presence-one/issues/230

where both PIR and mmWave seem to toggle on/off all the time.

I'm wondering if the device is restart-looping somehow.

EverythingSmartHome, 2025-09-09:
> Howdy, I tried the BOOT button on both my EP1 and EP Lite. Neither one blinked even after hold for roughly 30 seconds. Only thing that worked was the RESET and it all connected back to my network that I'm moving away from. Thank you,

It's slightly different now, you need to do the following (it was getting triggered accidently before with the 5s hold):

1. Press and hold boot for 1s
2. Release for 1s
3. Press and hold boot for 1s
4. Release for 1s
5. Press and hold boot for 5s
6. Should flash red 5s once done

closed 2026-09-16 by EverythingSmartHome