EverythingSmartHome/everything-presence-one #280: mmWave Radar C4001 (SEN0609) doesn't seem to work with EP1 board rev 1.4

olympia, 2025-08-25:
There’s no option for board revision 1.4 when doing USB flashing, so I selected 1.5 after swapping the original SEN0395 with a SEN0609 I just bought from the webshop. After flashing, mmWave detection doesn’t work. Does this indicate the sensor is dead, or that rev 1.4 boards simply aren’t supported (I couldn’t find any warning about this anywhere)?

saschachina, 2025-08-28:
> Absolutely, brilliant idea! I can’t seem to find the pinout diagram for the old board - do you happen to know it? If it works (and I think there’s a good chance it will), the next challenge will be figuring out how to mount the SEN0609 in the case and fit all the jumper wires.

I dont have the pinout im sorry, i dont find time at the moment to test it, i would just play around ;-)

EverythingSmartHome, 2026-05-31:
This is now covered in the current EP1 docs. The live hardware overview explicitly notes that SEN0395 is for rev 1.3 boards, while SEN0609 is for rev 1.5 and later boards: https://docs.everythingsmart.io/s/products/doc/hardware-overview-4UDfCXowuh

closed 2026-05-31 by EverythingSmartHome